ReFirmLabs/binwalk #676: external extractor

ojuanjo, 2024-03-22:
WARNING: Extractor.execute failed to run external extractor '7z e -y '%e'': 'module' object has no attribute 'DEVNULL', '7z e -y '%e'' might not be installed correctly
203           0xCB            LZMA compressed data, properties: 0xA2, dictionary size: 0 bytes, uncompressed size: 1610739727 bytes

WARNING: Extractor.execute failed to run external extractor '7z e -y '%e'': 'module' object has no attribute 'DEVNULL', '7z e -y '%e'' might not be installed correctly
8616          0x21A8          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 97728 bytes

WARNING: Extractor.execute failed to run external extractor '7z e -y '%e'': 'module' object has no attribute 'DEVNULL', '7z e -y '%e'' might not be installed correctly
70016         0x11180         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 2733760 bytes

WARNING: Extractor.execute failed to run external extractor 'unsquashfs -d 'squashfs-root' '%e'': 'module' object has no attribute 'DEVNULL', 'unsquashfs -d 'squashfs-root' '%e'' might not be installed correctly

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -le -d 'squashfs-root' '%e'': 'module' object has no attribute 'DEVNULL', 'sasquatch -p 1 -le -d 'squashfs-root' '%e'' might not be installed correctly

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -be -d 'squashfs-root' '%e'': 'module' object has no attribute 'DEVNULL', 'sasquatch -p 1 -be -d 'squashfs-root' '%e'' might not be installed correctly

WARNING: Extractor.execute failed to run external extractor 'mkdir 'squashfs-root' && mount -t squashfs '%e' 'squashfs-root'': 'module' object has no attribute 'DEVNULL', 'mkdir 'squashfs-root' && mount -t squashfs '%e' 'squashfs-root'' might not be installed correctly

mzpqnxow, 2024-03-23:
The errors say you need to install sasquatch and 7zip. Have you tried that?

see the deps.sh script in this repo

ojuanjo, 2024-03-25:
We apologize for not being able to provide this information:
1. The tools needed in the alarm are installed normally by me as well as they are under the path of my system environment variables,
2. When he reported an error, I tried to cp these tools to the same directory as the tools installed in the deps.sh file of Binwalk, which is also the path of my environment variables, but it didn't work

mzpqnxow, 2024-03-30:
Ok, well, let's look closely at the error

> WARNING: Extractor.execute failed to run external extractor 'mkdir 'squashfs-root' && mount -t squashfs '%e' 'squashfs-root'': 'module' object has no attribute 'DEVNULL', 'mkdir 'squashfs-root' && mount -t squashfs '%e' 'squashfs-root'' might not be installed correctly

If I recall correctly, DEVNULL is an import from subprocess, intended to be used with functions like popen, so you can instruct it to use /dev/null for the stdin, stdout or stderr descriptors

It sounds like the import is failing. Are you on a system that does not have a /dev/null device?

I can't imagine any Python version removing this but I also haven't looked at 3.12, 3.11 subprocess docs

mzpqnxow, 2024-03-31:
> We apologize for not being able to provide this information:
>
> 1. The tools needed in the alarm are installed normally by me as well as they are under the path of my system environment variables,
>
> 2. [...]

You realize that several of these tools have special patches applied, yes? When you say "installed normally", do you mean by following the steps in dep.sh, or do you mean using an OS package manager?

Either way, it seems like a waste of time trying to troubleshoot something that wasn't installed "properly". Sure, it's not rocket science cloning, patching and building a few packages- but you're bound to do something slightly differently than the script does, and then run into cases that mo other users have

Use the script and see if that helps

mzpqnxow, 2024-03-31:
Finally, you should probably switch to https://github.com/OSPG/binwalk as this repository seems to now be officially moribund

devttys0, 2024-09-23:
As @mzpqnxow pointed out, Binwalk relies on external utilities to perform extraction. In this case though, the `DEVNULL` errors make it look like perhaps a Python versioning issue? 

In either case, the Python version of binwalk is no longer supported. We recommend trying the latest Binwalk v3, which is Rust based. You will still need to have `sasquatch` installed to extract SquashFS images, but LZMA extraction is built-in and requires no external utilities.

closed 2024-09-23 by devttys0