ReFirmLabs/binwalk #688: AttributeError: module 'capstone' has no attribute 'CS_ARCH_ARM64'

1392081456, 2024-10-06:
System information：Linux kali 6.10.9-amd64 #1 SMP PREEMPT_DYNAMIC Kali 6.10.9-1kali1 (2024-09-09) x86_64 GNU/Linux
python information:   3.12.6

When I run binwalk, it appears：
Traceback (most recent call last):
  File "/usr/bin/binwalk", line 33, in <module>
    sys.exit(load_entry_point('binwalk==2.4.2', 'console_scripts', 'binwalk')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/binwalk", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/binwalk/__main__.py", line 24, in <module>
    import binwalk.modules
  File "/usr/lib/python3/dist-packages/binwalk/modules/__init__.py", line 3, in <module>
    from binwalk.modules.disasm import Disasm
  File "/usr/lib/python3/dist-packages/binwalk/modules/disasm.py", line 21, in <module>
    class Disasm(Module):
  File "/usr/lib/python3/dist-packages/binwalk/modules/disasm.py", line 60, in Disasm
    Architecture(type=capstone.CS_ARCH_ARM64,
                      ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'capstone' has no attribute 'CS_ARCH_ARM64'. Did you mean: 'CS_ARCH_ARM'?

I try to update binwalk，but it is the newest.  Then I followed the instructions to find the corresponding disasm.py and modify it。I modify the "ARM64" to "ARM",   just like：
ARCHITECTURES =[
Architecture(type=capstone.CS_ARCH_ARM,
mode=capstone.CS_MoDE_ARM,
endiannesS=capstone.CS_MODE_BIG_ENDIAN,
description="ARM executable code, 32-bit, big endian"),
Architecture(type=capstone.CS_ARCH_ARM,
mode=capstone.CS_MoDE_ARM,
endiannesS=capstone.CS_MODE_LITTLE_ENDIAN,
description="ARM executable code, 32-bit, little endian"),
Architecture(type=capstone.CS_ARCH_ARM,
mode=capstone.CS_MODE_ARM,
endianness=capstone.CS_MODE_BIG_ENDIAN,
description="ARM executable code, 64-bit, big endian"),
Architecture(type=capstone.CS_ARCH_ARM,
mode=capstone.CS_MODE_ARM,
romendiannesS=capstone.CS_MODE_LITTLE_ENDIAN,
Devicesdescription="ARM executable code, 64-bit, little endian"),

binwalk is then ready to run。

I would like to know why this is, and does my modification affect binwalk?

1392081456, 2024-10-06:
Do I need to uninstall it and reinstall it？

devttys0, 2024-10-06:
It looks like this error is due to a change to the Capstone API.

Your patch will only affect CPU opcode detection (`--opcodes`) for ARM: you won't be able to detect ARM64, and/or ARM32 may be reported as ARM64.

Note that current development efforts are focused on [Binwalkv3](https://github.com/ReFirmLabs/binwalk/tree/binwalkv3), which no longer supports the `--opcodes` argument. As a replacement I recommend the [cpu_rec](https://github.com/trou/cpu_rec_rs) tool, which is much better for identifying raw CPU instructions in a file.

closed 2024-10-06 by devttys0

1392081456, 2024-10-06:
Thank you for your reply！ The version of capstone I found is (6.0.0a1)，which version is more suitable?

CdcEta, 2025-01-11:
Have you solved the problem? I got this wrong, too.

Yo1o-sir, 2025-04-02:
So how to solve this problem?

LoveSunday7, 2025-04-20:
这老外有点无语了，把问题发出来不解决

LoveSunday7, 2025-04-23:
这老外有点牛逼哈，我也改好了

liudonghua123, 2025-05-28:
I have the same error when using binwalk on kali.

```
┌──(liudonghua㉿LDH-DESKTOP)-[~]
└─$ binwalk '/mnt/d/Downloads/running/run.exe'
Traceback (most recent call last):
  File "/usr/bin/binwalk", line 2, in <module>
    from binwalk.__main__ import main
  File "/usr/lib/python3/dist-packages/binwalk/__main__.py", line 24, in <module>
    import binwalk.modules
  File "/usr/lib/python3/dist-packages/binwalk/modules/__init__.py", line 3, in <module>
    from binwalk.modules.disasm import Disasm
  File "/usr/lib/python3/dist-packages/binwalk/modules/disasm.py", line 21, in <module>
    class Disasm(Module):
  File "/usr/lib/python3/dist-packages/binwalk/modules/disasm.py", line 60, in Disasm
    Architecture(type=capstone.CS_ARCH_ARM64,
                      ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'capstone' has no attribute 'CS_ARCH_ARM64'. Did you mean: 'CS_ARCH_ARM'?

```

And I have tried the following two methods. Both of them works for me.

1. install the latest binwalk v3 via `cargo install binwalk`
2. install the latest [capstone](https://pypi.org/project/capstone/) v5 via `pip install capstone` or `sudo apt install python3-capstone`.

<details><summary>Details</summary>

```
┌──(liudonghua㉿LDH-DESKTOP)-[~/ctf]
└─$ cargo install binwalk
......
┌──(liudonghua㉿LDH-DESKTOP)-[~/ctf]
└─$ which binwalk
/home/liudonghua/.cargo/bin/binwalk

┌──(liudonghua㉿LDH-DESKTOP)-[/mnt/d/Downloads/running]
└─$ binwalk run.exe

                                            /mnt/d/Downloads/running/run.exe
------------------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
------------------------------------------------------------------------------------------------------------------------
0                                  0x0                                Windows PE binary, machine type: Intel x86
199872                             0x30CC0                            PNG image, total size: 2885 bytes
202760                             0x31808                            PNG image, total size: 5545 bytes
230692                             0x38524                            PNG image, total size: 15729 bytes
429056                             0x68C00                            ZIP archive, file count: 2, total size: 259512
                                                                      bytes
------------------------------------------------------------------------------------------------------------------------

Analyzed 1 file for 85 file signatures (187 magic patterns) in 12.0 milliseconds

┌──(liudonghua㉿LDH-DESKTOP)-[/mnt/d/Downloads/running]
└─$ pip install capstone
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
[... 115 lines cut ...]
    -Z, --lzma                   Scan for raw LZMA compression streams
    -P, --partial                Perform a superficial, but faster, scan
    -S, --stop                   Stop after the first result

General Options:
    -l, --length=<int>           Number of bytes to scan
    -o, --offset=<int>           Start scan at this file offset
    -O, --base=<int>             Add a base address to all printed offsets
    -K, --block=<int>            Set file block size
    -g, --swap=<int>             Reverse every n bytes before scanning
    -f, --log=<file>             Log results to file
    -c, --csv                    Log results to file in CSV format
    -t, --term                   Format output to fit the terminal window
    -q, --quiet                  Suppress output to stdout
    -v, --verbose                Enable verbose output
    -h, --help                   Show help output
    -a, --finclude=<str>         Only scan files whose names match this regex
    -p, --fexclude=<str>         Do not scan files whose names match this regex
    -s, --status=<int>           Enable the status server on the specified port

┌──(liudonghua㉿LDH-DESKTOP)-[/mnt/d/Downloads/running]
└─$
```

</details>