ReFirmLabs/binwalk #695: Questions for V3

ultimateshadsform, 2024-10-14:
Will V3 also depend on external tools like unsquashfs and cpio etc or will it be inbuilt/an addon in rust?

It would be nice to not have to install any external tools to extract. Just a thought.

devttys0, 2024-10-14:
External dependency hell is a real problem, for sure:

- They're a PITA to install, even for me, who is intimately familiar with which tools are required
- It limits non-Linux OS support, as many external tools are Linux-only
- Installation of external tools even varies between different Linux distros
- External tools may have bugs that have to be patched upstream

The advantages of using external tools are:

- Easy to rapidly add extraction support for extraction of various file formats
- Many standard tools are well vetted and support all the edge cases/intricacies of complex file formats
- Binwalk is owned by Microsoft, and for legal reasons they don't want any third-party encryption keys hosted in their repositories; this means that to support decryption of firmware that uses known encryption keys, an external tool must be used

So while I 100% agree with you, v3 will still require external tools. However, it was built from the ground up with support for internal extractors, with the intent of gradually replacing as many external extractors as possible with internal ones moving forward. Currently v3 supports extraction of 53 file types, 23 of those use internal extractors.

There is a Dockerfile included with v3 that builds and installs Binwalk and all external tools. Though personally I prefer having Binwalk installed locally, this does allow you to get everything set up with a single docker command regardless of your host OS and without cluttering your local system with all the additional tools.

There are dependency build scripts in the works to simplify local installation of the external tools, but these are currently a work-in-progress and will need to be tweaked to support various Linux distros.

ultimateshadsform, 2024-10-14:
Thank you for the information.

closed 2024-10-31 by devttys0