ReFirmLabs/binwalk #697: Add Cargo.lock file

KSJ2000, 2024-10-15:
Please add the Cargo.lock file so that when building the nix derivation for the rust implementation of binwalk, it will not needed to define each dependency and it's hash, because the cargoHash will be used.
https://nixos.org/manual/nixpkgs/stable/#rust
It'd be great if you could also add a nix derivation for a fully reproducible build of binwalk, since what is from the Dockerfile for sure isn't, since the answer is no to the question: will this result in the same build inputs in half a year, or maybe 2 years.
https://github.com/ReFirmLabs/binwalk/blob/d0c89d855bee7cb8e3bfe7424db2e5c59ab88ebd/Dockerfile#L6

devttys0, 2024-10-15:
Added Cargo.lock to the git repo. I like the nix idea, I'll leave this issue open until that is implemented. Thanks!

devttys0, 2024-10-31:
https://github.com/NixOS/nixpkgs/issues/350300

KSJ2000, 2024-11-01:
Do I not understand this correctly?
https://github.com/ReFirmLabs/binwalk/blob/e0a4434159f22ff926c1baa3872d93a5f8dbf87e/dependencies/README.md?plain=1#L3
https://github.com/ReFirmLabs/binwalk/blob/e0a4434159f22ff926c1baa3872d93a5f8dbf87e/dependencies/pip.sh#L9
https://github.com/ReFirmLabs/binwalk/blob/e0a4434159f22ff926c1baa3872d93a5f8dbf87e/dependencies/src.sh#L5
Are those not needed for runtime?
Like... where are the dependencies in the buildInputs...
Just because V3 builds, doesn't mean it also has all the necessary dependencies it needs to work in every scenario.
Some dependencies don't even exist in nixpkgs.
I'm trying to get them merged, but it will take some more time until everything is done.
https://github.com/NixOS/nixpkgs/pull/349055
https://github.com/NixOS/nixpkgs/pull/348768

devttys0, 2024-11-27:
https://github.com/NixOS/nixpkgs/pull/357991

closed 2024-11-27 by devttys0