ReFirmLabs/binwalk #702: Lint issue

banditopazzo, 2024-10-19:
I created PR #701 and I wanted to add also a lint check in the github wokflow but I noticed that there are too many issue I can solve myself in a single PR. Running the standard `clippy` to perform check it reports **747** issues in the source code.

The specific command I used is:
```shell
cargo clippy -- -D warnings
```

I can solve all the issues and make a PR, but expect a giant boring review. Probably it's better if you make the changes yourself.
Let me know if you want some help.

After this first iteration I can add another github workflow for the lint

devttys0, 2024-10-19:
Thanks! PR merged, I'm knocking out the lint issues in https://github.com/ReFirmLabs/binwalk/tree/lint-fixes.

devttys0, 2024-10-21:
Clippy now reports no issues in the source code.

closed 2024-10-21 by devttys0

banditopazzo, 2024-10-21:
awesome