ReFirmLabs/binwalk #703: Extract with docker broken on master branch

Flo354, 2024-10-19:
Hello,

The docker image doesn't work with the "-e" argument. It is working perfectly with tag "version-2.3.4"

```bash
<> docker run -it --rm -v "$(pwd):/home/appuser" refirmlabs/binwalk -e "file.bin"
[2024-10-19T13:43:56Z ERROR binwalk::binwalk] Failed to create symlink /home/appuser/extractions/file.bin -> /home/appuser/file.bin: Permission denied (os error 13)
thread 'main' panicked at src/main.rs:81:6:
Binwalk initialization failed: BinwalkError
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[1]    76653 exit 101   docker run -it --rm -v "$(pwd):/home/appuser" refirmlabs/binwalk -e 
```

devttys0, 2024-10-19:
From the `permission denied` error, it looks like a file permission problem; specifically, the user ID in the docker container (`1337` by default) does not have write access to your $PWD.

You can check this by making $PWD `/tmp`, which is writable by everyone, and running the same command:

```
cp file.bin /tmp/file.bin
cd /tmp
sudo docker run -it --rm -v "$(pwd):/home/appuser" refirmlabs/binwalk -e "file.bin"
```

aioue, 2024-10-19:
Adjacent issue, solved by `sudo su -` on mac, and then running:

`# docker run -it -u 0 --rm  -v "$(pwd):/home/appuser" refirmlabs/binwalk:version-2.3.4 --run-as=root -Me "BIOS.BIN"`

Flo354, 2024-10-21:
Hello,

so, the solution with `/tmp` folder works. The solution with `-u 0` works same. But in my opinion there is an issue. 

1/ It was working smoothly on version 2.3.4. I had no such arguments to put.

2/ look at the `extractions` folder after the execution
[image]
The program is creating a symbolic link I don't really understand why, and it is linked to the docker container, so obviously not found.

Regards,

Flo354, 2024-10-21:
for the user, ok, I just did it with -u 1000 and that's solving most of it.
Now, for the symbolic link, isn't it possible to add an argument allowing the program to remove the symbolic link at the end?

devttys0, 2024-10-22:
I just merged some Dockerfile and minor code updates into the master branch which should fix these issues:

1. If the `BINWALK_RM_EXTRACTION_SYMLINK` environment variable is set, then the symlink in question above will be deleted at the end of analysis.
2. The `BINWALK_RM_EXTRACTION_SYMLINK` environment variable is set automatically in the docker container, so when running via docker there is nothing else you need to do or add.
3. The default user ID of the docker container is now `1000`, so the `-u 1000` docker argument is no longer necessary if your local user ID is already `1000`. If your user ID is *not* `1000`, you can still tell docker to run with your user id: `-u $(id -u $(whoami))`.

I think these should be sane defaults for the most common cases while still providing user-definable flexibility for other cases.

> [!NOTE]
> The working directory for the docker container is now `/analysis`, *not* `/home/appuser`, so you will have to change your docker `-v` argument to:
>
>```
> sudo docker run -it ---rm -v "$(pwd):/analysis" refirmlabs/binwalk -e "file.bin"
>```

I've updated the Wiki with some notes on this as well.

closed 2024-10-22 by devttys0

Flo354, 2024-10-23:
Thank you, it works.

If it can help, here are two aliases I set to ease me in the task

```bash
alias binwalk="BINWALK_RM_EXTRACTION_SYMLINK=1 docker run -it --rm -u $(id -u $(whoami)) -v .:/analysis refirmlabs/binwalk"

alias binwalk_update="docker pull refirmlabs/binwalk"
```