ReFirmLabs/binwalk #704: What about the sasquatch build depency?

E3V3A, 2024-10-21:
I see from your "building from sources" page, that you require the Debian package [sasquatch](https://github.com/onekey-sec/sasquatch). 

What exactly is this used for? 

I would like to build this for windows via MinGW/MSYS etc, so already have all the pieces except that annoyance. (Yeah, sorry, but Docker is for ignorant kids.)

devttys0, 2024-10-21:
It is executed by Binwalk to extract SquashFS file systems. It is not a required dependency unless you want Binwalk to auto-extract SquashFS file systems for you (it will still identify and report their existence if it finds one).

Sasquatch was originally a set of patches that I wrote for the `unsquashfs` utility many years ago, allowing it to handle lots of different non-standard SquashFS implementations found in the wild. My repository for it has not been updated in a long time, but  one-key has a more [actively maintained version](https://github.com/onekey-sec/sasquatch/).

E3V3A, 2024-10-21:
Funny, I just found my own fork of your version. I haven't been here for ages! 😄  The question is, if there is still any useful development on squashfs? It seemed to work fine many years ago, so...

devttys0, 2024-10-21:
It looks like they've at least added support for SquashFSv4 big-endian (official SquashFSv4 is little endian only, and IIRC my sasquatch patches assumed the same). I have seen this once in the wild.

closed 2024-10-21 by devttys0