ReFirmLabs/binwalk #714: Add support for BTRFS Fillesystem

shiren00, 2024-10-25:
We would appreciate it if anyone with additional materials or actual disk image files related to this could share them :D

- https://btrfs.readthedocs.io/en/latest/dev/dev-btrfs-design.html

devttys0, 2024-10-29:
WIP: https://github.com/ReFirmLabs/binwalk/tree/btrfs_support

devttys0, 2024-10-29:
BTRFS identification is working and merged into the master branch. Unfortunately, there doesn't seem to be an easy existing solution to extracting the files from a BTRFS image (the btrForensics tool above is interactive, not automated, and requires knowing the inode number of each file you want to extract; it also fails on BTRFS images created with default `mkfs.btrfs` settings and doesn't look like it's been updated in several years).

Since writing a custom extractor would require a good deal more time, and I'm trying to push a stable release of Binwalk this week, I'm closing this issue and I've added issue #724 to specifically write a BTRFS extractor.

closed 2024-10-29 by devttys0