ReFirmLabs/binwalk #716: Add support for APFS Fillesystem

shiren00, 2024-10-25:
We would appreciate it if anyone with additional materials or actual dump files related to this could share them :D

- https://en.wikipedia.org/wiki/Apple_File_System
- https://developer.apple.com/support/downloads/Apple-File-System-Reference.pdf

devttys0, 2024-10-26:
WIP: https://github.com/ReFirmLabs/binwalk/tree/apfs_support

devttys0, 2024-10-28:
APFS support is functional and merged into the master branch.

Note that tsk_recover was not able to actually extract any of the APFS images that I created (using both Linux utilities and the OSX disk manager on a MacBook). The [latest version of 7zip](https://www.7-zip.org/a/7z2408-linux-x86.tar.xz) does work though, and has been added as a new external dependency.

closed 2024-10-28 by devttys0

shiren00, 2024-10-29:
Thank you very much!!