ReFirmLabs/binwalk #740: Out of memory error when trying to analyze large dump

FearlessArmada, 2024-11-06:
```
[2024-11-06T13:29:13Z ERROR binwalk::common] Failed to read file /home/user/documents/boox/backup/flash.bin.lun0 into memory: out of memory
Analyzed 1 file for 85 file signatures (187 magic patterns) in 2.0 milliseconds
```
The file is very large to be fair.

```115G	flash.bin.lun0```
Is there a workaround other than swap, like splitting up large files into chunks and analyzing those? System memory is 64GB, with a 4GB swap.

devttys0, 2024-11-06:
Yes, this is a [limitation](https://github.com/ReFirmLabs/binwalk/wiki/Limitations) of the current implementation; files are read into memory in their entirety so you need as much free memory as the size of the file(s) you're analyzing. 

You can split the file into multiple smaller files and analyze them individually, though this may cause binwalk to think that some otherwise valid signatures are invalid, depending on the contents of the `flash.bin.lun0` file. Example: if the entire file is a single FAT32 file system and you split the file in half, binwalk will think the FAT32 is invalid since the FAT32 image reports that it is 115GB in size, but the file you are scanning is only 57.5GB. If you set the `RUST_LOG=debug` environment variable, you would see the log messages report something along the lines of "Found fat magic bytes...fat extends beyond EOF, ignoring".

FearlessArmada, 2024-11-06:
I

closed 2024-11-06 by FearlessArmada

mid-kid, 2024-11-12:
This is a bit disappointing, especially given how the 2.x series had no problems supporting it...

racerxdl, 2025-01-01:
That's really disappointing. I could use previous version directly on disks and such and now I can't :-1: