ReFirmLabs/binwalk #741: Preserve carved files when extracting in binwalk3

jandress, 2024-11-07:
In binwalk v2.3.x, extracting files preserves the carved files unless -r is passed to delete them. Also we have the -z option to carve files without extracting them. Binwalk 3.1.x, as far as I can tell, doesn't seem to have this functionality and automatically cleans up all the intermediate files, which is not universally desirable. Is there a way to either keep these files or carve them without extraction in binwalk3?

1nikolas, 2024-11-19:
Yes please add this

devttys0, 2024-11-21:
WIP: https://github.com/ReFirmLabs/binwalk/tree/keep_carved_files

devttys0, 2024-11-21:
A new command line option, `--carve`, has been added to the master branch which preserves carved files during extraction.

Note that this works slightly differently than it did in Binwalkv2:

- In v2, `--carve` disabled execution of extractors; it does *not* disable extractors in v3
- Carved file names will be in the format: `<original file name>_<file offset>_<signature type>.raw`
- Unknown chunks of data will also be carved to disk; these carved file names will have the `<signature type>` of `unknown`

closed 2024-11-21 by devttys0