ReFirmLabs/binwalk #742: [v3.1.0] [Macos] Failed to execute command sasquatch

douniwan5788, 2024-11-09:
```
[2024-11-09T08:49:59Z ERROR binwalk::extractors::common] Failed to execute command sasquatch["xxx/extact_HS8545M5_V500R020C00SPC200B459/extractions/flash_rootfs.extracted/2C1B054/squashfs_2C1B054.sqsh"]: No such file or directory (os error 2)
[2024-11-09T08:49:59Z ERROR binwalk::extractors::common] Failed to spawn external extractor for 'squashfs' signature: No such file or directory (os error 2)
[2024-11-09T08:49:59Z ERROR binwalk::extractors::common] Failed to execute command sasquatch["xxx/extact_HS8545M5_V500R020C00SPC200B459/extractions/flash_rootfs.extracted/2C1B054/squashfs_2C1B054.sqsh"]: No such file or directory (os error 2)
[2024-11-09T08:49:59Z ERROR binwalk::extractors::common] Failed to spawn external extractor for 'squashfs' signature: No such file or directory (os error 2)
```

brainstorm, 2024-11-12:
Simply install https://github.com/devttys0/sasquatch in your system.

devttys0, 2024-11-14:
@brainstorm is correct, this error simply means that `sasquatch` is not installed on your system. 

Binwalk relies on several external utilities to perform extraction, these can be displayed by running `binwalk --list`.

FYI, there is a more recently maintained version of sasquatch [here](https://github.com/onekey-sec/sasquatch/), but I don't know if Mac is supported.

closed 2024-11-14 by devttys0