ReFirmLabs/binwalk #757: binwalk on mac

hedgehogdrfrink, 2024-11-21:
Hi,

Looking to use binwalk on macos, I tried to use brew install binwalk though seems to be running into some issues.

user@users-MacBook-Pro Downloads % binwalk -D=".*" file.zip
error: unexpected argument '-D' found

  tip: to pass '-D' as a value, use '-- -D'

Usage: binwalk [OPTIONS] [FILE_NAME]

For more information, try '--help'.
user@users-MacBook-Pro Downloads % binwalk --help
Analyzes data for embedded file types

Usage: binwalk [OPTIONS] [FILE_NAME]

Arguments:
  [FILE_NAME]  Path to the file to analyze

Options:
  -L, --list                   List supported signatures and extractors
  -q, --quiet                  Supress output to stdout
  -v, --verbose                During recursive extraction display *all* results
  -e, --extract                Automatically extract known file types
  -M, --matryoshka             Recursively scan extracted files
  -a, --search-all             Search for all signatures at all offsets
  -E, --entropy                Plot the entropy of the specified file
  -l, --log <LOG>              Log JSON results to a file
  -t, --threads <THREADS>      Manually specify the number of threads to use
  -x, --exclude <EXCLUDE>...   Do no scan for these signatures
  -y, --include <INCLUDE>...   Only scan for these signatures
  -C, --directory <DIRECTORY>  Extract files/folders to a custom directory [default: extractions]
  -h, --help                   Print help
  -V, --version                Print version
user@users-MacBook-Pro Downloads % man binwalk
No manual entry for binwalk
user@users-MacBook-Pro Downloads % 

How would I compile binwalk from source on a mac?

devttys0, 2024-11-22:
It looks like the brew install worked and Binwalk is installed. The `-D` option is not available in Binwalkv3. 

An alternate to `-D` / `--dd` in Binwalkv3 is `--carve`, which does the same thing as `-D='.*'` did in Binwalkv2. It was just merged into the master branch today though, so unless brew is pulling from the master branch, you'll have to compile from source. There are [instructions](https://github.com/ReFirmLabs/binwalk/wiki/Compile-From-Source) on the Wiki to do this, though I have not done any testing on OSX.

hedgehogdrfrink, 2024-11-22:
Thank you, was able to compile and build for mac, the ubuntu dependencies weren't needed

devttys0, 2024-11-22:
The `--carve` option will carve both known and unknown data blobs from a file to their own separate files. Example:

[image]

Here there are two known "data blobs", a uImage Firmware Image and a SquashFS file system, with some unknown data that resides in between them.

If, however, there are no identified signatures, then there is really nothing to "carve" out of the file, which I suspect is the case with your mkv file.

hedgehogdrfrink, 2024-11-22:
Tried with binwalk v2.3.4, was able to get far more, ended up extracting about 159 GB of files from an .mkv, here's the output from the master repo and binwalk v2.3.4 respectively with some identifying info replace with ...

From master repo of binwalkv3,
DECIMAL                            HEXADECIMAL                        DESCRIPTION
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
41138                              0xA0B2                             JPEG image, total size: 24949 bytes
83515                              0x1463B                            Copyright text: "Copyright belongs to the Creator."
150537                             0x24C09                            Copyright text: "Copyright..."
256568                             0x3EA38                            Copyright text: "Copyright..."
322169                             0x4EA79                            Copyright text: "Copyright..."
552357                             0x86DA5                            Copyright text: "Copyright..."
584869                             0x8ECA5                            Copyright text: "Copyright..."
592226                             0x90962                            Copyright text: "Copyright..."
832730                             0xCB4DA                            Copyright text: "Copyright..."

binwalk v2.3.4 results with --dd='.*' [mkv File]
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             EBML file, Matroska data
4734          0x127E          Copyright string: "Copyright 2013-2017 (c) Multicoreware, Inc - http://x265.org - options: cpuid=1173503 frame-threads=3 numa-pools=8 wpp no-pmode "
41138         0xA0B2          JPEG image data, JFIF standard 1.01
83515         0x1463B         Copyright string: "Copyright belongs to the Creator."
150537        0x24C09         Copyright string: "Copyright..."
150763        0x24CEB         Copyright string: "Copyright..."
322169        0x4EA79         Copyright string: "Copyright..."
322411        0x4EB6B         Copyright string: "Copyright section for the font trademark attribution notices."
552357        0x86DA5         Copyright string: "Copyright..."
552535        0x86E57         Copyright string: "Copyright..."
584869        0x8ECA5         Copyright string: "Copyright..."
592226        0x90962         Copyright string: "Copyright..."
614063        0x95EAF         Object signature in DER format (PKCS header length: 4, sequence length: 5238
614207        0x95F3F         Certificate in DER format (x509 v3), header length: 4, sequence length: 700
614911        0x961FF         Certificate in DER format (x509 v3), header length: 4, sequence length: 938
615853        0x965AD         Certificate in DER format (x509 v3), header length: 4, sequence length: 1026
616883        0x969B3         Certificate in DER format (x509 v3), header length: 4, sequence length: 1311
617038        0x96A4E         Digi International firmware, load address: 0x204D6963, entry point: 0x66742053,
643831        0x9D2F7         MySQL MISAM compressed data file Version 6
832730        0xCB4DA         Copyright string: "Copyright..."
832956        0xCB5BC         Copyright string: "Copyright..."
842330        0xCDA5A         Copyright string: "Copyright..."
5651730       0x563D12        Zlib compressed data, best compression
7186273       0x6DA761        Zlib compressed data, best compression
7289435       0x6F3A5B        Zlib compressed data, best compression
7911886       0x78B9CE        Zlib compressed data, best compression
8005726       0x7A285E        Zlib compressed data, best compression
9206249       0x8C79E9        Zlib compressed data, best compression
9257573       0x8D4265        Zlib compressed data, best compression
11594045      0xB0E93D        Zlib compressed data, best compression
11656012      0xB1DB4C        Zlib compressed data, best compression
13535275      0xCE882B        Zlib compressed data, best compression
13535458      0xCE88E2        Zlib compressed data, best compression
17259099      0x1075A5B       Zlib compressed data, best compression
18758692      0x11E3C24       Zlib compressed data, best compression
20986228      0x1403974       Zlib compressed data, best compression
21040761      0x1410E79       Zlib compressed data, best compression
21392306      0x1466BB2       MySQL MISAM index file Version 2
21509856      0x14836E0       Zlib compressed data, best compression
21509899      0x148370B       Zlib compressed data, best compression
21518580      0x14858F4       Zlib compressed data, best compression
22395374      0x155B9EE       Zlib compressed data, best compression
22395416      0x155BA18       Zlib compressed data, best compression
31346602      0x1DE4FAA       Zlib compressed data, best compression
31352553      0x1DE66E9       Zlib compressed data, best compression
31699040      0x1E3B060       Zlib compressed data, best compression
31850876      0x1E6017C       Zlib compressed data, best compression
31853908      0x1E60D54       Zlib compressed data, best compression
32212650      0x1EB86AA       Zlib compressed data, best compression
32212692      0x1EB86D4       Zlib compressed data, best compression
[... cut ...]
430035021     0x19A1D04D      Zlib compressed data, best compression
430061820     0x19A238FC      Zlib compressed data, best compression
435870275     0x19FADA43      Zlib compressed data, best compression
435870318     0x19FADA6E      Zlib compressed data, best compression
435895230     0x19FB3BBE      Zlib compressed data, best compression
435915954     0x19FB8CB2      Zlib compressed data, best compression
441949444     0x1A579D04      Zlib compressed data, best compression
441949486     0x1A579D2E      Zlib compressed data, best compression

hedgehogdrfrink, 2024-11-22:
I'm going to stick with Binwalk v2.3.4 for now, though honestly thank you for this amazing software, it's inspired me to get into learning rust in hopes I'll get good enough to contribute to this project.

closed 2024-11-24 by devttys0