ReFirmLabs/binwalk #797: AttributeError: module 'capstone' has no attribute 'CS_ARCH_ARM64'

FloocX, 2024-12-07:
My ubuntu is 20.06. While pwngdb is installed, there is such wrong.
No pwngdb installed, it can works.
binwalk is the latest version

```
Traceback (most recent call last):
  File "/usr/bin/binwalk", line 26, in <module>
    import binwalk.modules
  File "/usr/lib/python3/dist-packages/binwalk/modules/__init__.py", line 3, in <module>
    from binwalk.modules.disasm import Disasm
  File "/usr/lib/python3/dist-packages/binwalk/modules/disasm.py", line 21, in <module>
    class Disasm(Module):
  File "/usr/lib/python3/dist-packages/binwalk/modules/disasm.py", line 60, in Disasm
    Architecture(type=capstone.CS_ARCH_ARM64,
AttributeError: module 'capstone' has no attribute 'CS_ARCH_ARM64'
```

closed 2024-12-07 by devttys0

FloocX, 2024-12-07:
> The latest release of Binwalk has been re-written in Rust, and no longer uses the capstone library.
> 
> For identifying CPU architectures, I'd highly recommend [cpu_rec](https://github.com/trou/cpu_rec_rs).

but from the cpu_rec,i it looks like this tool has been added to the dependencies?

devttys0, 2024-12-09:
> it looks like this tool has been added to the dependencies?

Not sure which tool (Binwalk? cpu_rec?) you mean here, so I'll try to cover all the bases:

1. Binwalk no longer uses the capstone library
2. Binwalk does not depend on cpu_rec
3. cpu_rec does not depend on Binwalk
4. I could be wrong, but I don't believe that cpu_rec depends on the capstone library