ReFirmLabs/binwalk #801: 7zip is safe?

githubDiversity, 2024-12-16:
I noticed today that the dependencies installer script (~/binwalk/dependencies/ubuntu.sh ) installs 7zip.

when I then want to see which version that was;
# sudo apt list 7z*
Listing... Done
7zip/stable 22.01+dfsg-8+deb12u1 amd64

Is that a version with the vulnerability already patched?
this article mentions version 24.07 as being patched but does not mention 22.01
https://linuxsecurity.com/news/security-vulnerabilities/7-zip-urgent-update-code-execution-flaw

or am I overlooking something else?

Lerenelmax, 2024-12-16:
[image]

githubDiversity, 2024-12-16:
? this is a joke yes? it must be. if not that would be the first time in history a possible security issue is being responded to in such a manner

devttys0, 2024-12-16:
Not sure what the picture is for, but that is a question for whoever maintains the 7z package for your Linux distro @githubDiversity. Binwalk cannot guarantee the safety of external utilities. 

If you are concerned about it, you can always [install 7z manually](https://7-zip.org/download.html), but of course then you will be responsible for manually keeping 7z up-to-date against any new security threats.

closed 2024-12-16 by devttys0