ReFirmLabs/binwalk #805: Support for big endian detection in sqashfs (and possibly other signatures)

1nikolas, 2024-12-23:
It seems that unlike v2, v3 doesn't detect big endian Squashfs portions. I don't know if this applies to other signatures but I stumbled upon this example, I can send the firmware file if needed. Thanks

[image]

devttys0, 2024-12-24:
Big endian SquashFS is supported, so perhaps it is just this particular signature that isn't recognized. If you can provide the firmware file I can take a look at it.

1nikolas, 2024-12-24:
[101AZG0C0.bin.zip](https://github.com/user-attachments/files/18240683/101AZG0C0.bin.zip)

This is the file. I think this did not extract with unsquashfs I had to use the other one (sasquatch or how it's called).

devttys0, 2024-12-24:
This was an issue with parsing version 2 of the SquashFS superblock structure, it has been fixed in the master branch:

[image]

closed 2024-12-24 by devttys0