ReFirmLabs/binwalk #816: Doesn't detect zip header with rust rewrite

theoparis, 2025-01-06:
Relevant binary file: https://download.asrock.com/TSD/Afulnx.zip

Rust rewrite:
```
root@tinted:~# which binwalk
/root/.cargo/bin/binwalk
root@tinted:~# binwalk Afulnx.zip
Analyzed 1 file for 101 file signatures (226 magic patterns) in 4.0 milliseconds
```

python binwalk from debian repositories:
```
root@tinted:~# /usr/bin/binwalk Afulnx.zip

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
122           0x7A            Zip archive data, at least v1.0 to extract, name: AfuLnx64/
161           0xA1            Zip archive data, at least v1.0 to extract, name: AfuLnx64/AfuLnx64/
209           0xD1            Zip archive data, at least v2.0 to extract, compressed size: 351822, uncompressed size: 1103728, name: AfuLnx64/AfuLnx64/afulnx_64
352088        0x55F58         Zip archive data, at least v2.0 to extract, compressed size: 1930, uncompressed size: 5194, name: AfuLnx64/AfuLnx64/readme.txt
```

devttys0, 2025-01-07:
Binwalk is internally marking the ZIP file as invalid because, well, it is. :)  This ZIP archive appears to be truncated, prematurely ending part-way through one of the ZIP central-directory headers, and is missing the end-of-central-directory header entirely.

That said, extraction of at least some of the ZIP archive files is possible. I'll add support for this, though be aware that some data may be missing/corrupt.

devttys0, 2025-02-02:
Latest master branch now supports truncated ZIP files:

[image]

closed 2025-02-02 by devttys0