ReFirmLabs/binwalk #821: unable to identify firmware

blshkv, 2025-01-26:
Hello!

Firstly, thanks for the great tool!

I'm trying to decompile a firmware for coospo cs500 bike computer and a bit stack. I'm not able to identify any blocks manually, and
binwalk fails too.

Would you have any ideas, please?

The binary is uploaded here:
https://filebin.net/qoawwg8hla6firo8
The link will expire in 5 days, let me know if you want me to re-upload somewhere. The file is 800k only.

devttys0, 2025-03-05:
Sorry, the 5 days have obviously passed at this point. :)  For something that small I expect it is probably raw code, so there's likely nothing to extract - you'd have to disassemble/decompile the code in Ghidra/Binja/IDA/etc. Have you tried the `cpu_rec_rs` tool? It's quite good at detecting code and reporting the CPU architecture.

blshkv, 2025-03-07:
I think you are right, it's just a compiled binary with nothing to unpack. I found some samples, see the following URL https://devzone.nordicsemi.com/f/nordic-q-a/90048/create-zip-firmware-package-for-dfu-over-bluetooth and two attachments: dfu_application.zip peripheral_uart_smp.zip.  cpu_rec_corpus detected "Whole file | ARMhf"

cs500 firmware is reported as "Whole file | xmos_xs2a", fyi.

So I'll close this issue, binwalk is a wrong tool. 

Thanks a lot for your hints!

closed 2025-03-07 by blshkv