ReFirmLabs/binwalk #826: How to extract encrypted zip file?

abcfy2, 2025-02-10:
When file contains encrypted zip file, binwalk will extract empty file contents from the zip file.

But how to extract the original encrypted zip file instead ?

Then I can use `unzip` or `7z` to extract it with password.

Or is there any way to pass password to binwalk to extract the zip file?

pUrGe12, 2025-02-10:
As far as I know, in the extracted directory you will find the zip files with the normal extract mode itself. Then you can use `unzip` or `gzip` or `7z` to extract it further (check file type with `file`)

abcfy2, 2025-02-11:
> As far as I know, in the extracted directory you will find the zip files with the normal extract mode itself. Then you can use `unzip` or `gzip` or `7z` to extract it further (check file type with `file`)

no. I can see the `zipbin` file during extracting, but without password, all files in zip will extract with empty content, and the `zipbin` file finally deleted.

pUrGe12, 2025-02-11:
I see. What happens if you try extracting that file using `dd`? Just to verify that if it's a binwalk issue or something else.

abcfy2, 2025-02-11:
> I see. What happens if you try extracting that file using `dd`? Just to verify that if it's a binwalk issue or something else.

Yeah. `dd` is working:

```sh
dd if=1.mp4 of=clean.zip bs=1M iflag=skip_bytes skip=21668412
7z x clean.zip
# input password
# DONE
```

devttys0, 2025-03-05:
Binwalk does not support passing arbitrary arguments/inputs to extraction tools, and thus cannot provide a custom password for encrypted ZIPs and the like. As a work-around, the `--carve` option will carve the original ZIP file to disk during extraction, and you could then manually decrypt it.

closed 2025-03-05 by devttys0