ReFirmLabs/binwalk #836: Failed to execute command 7zz

vritzenthaler, 2025-03-09:
```
[2025-03-09T19:33:36Z ERROR binwalk::extractors::common] Failed to execute command 7zz["x", "-y", "-o.", "-p''", "/home/valentin/Documents/extractions/cutie.png.extracted/8702/zip_8702.bin"]: No such file or directory (os error 2)
[2025-03-09T19:33:36Z ERROR binwalk::extractors::common] Failed to spawn external extractor for 'zip' signature: No such file or directory (os error 2)

```

Hello, 

I recently installed binwalk by hand on my system (20.04.1-Ubuntu) and I got this error when running my first extract. I tried to find out what is 7zz, but it seems like it is not part of the 7zip suit. I'm just stuck right now. If someone has any idea how to solve I'm really interested for a solution. 

Thank you in advance.

taylor00541, 2025-03-10:
Related to #835 

It is worth noting that this also happens in the docker version: `7zz` is not installed, and in https://github.com/ReFirmLabs/binwalk/blob/master/dependencies/src.sh script `7zzs` is installed. So the (docker) installation script should be fixed, or the code (#819).

waiwai24, 2025-03-16:
@vritzenthaler 
On Debian/Ubuntu-based systems, install p7zip-full:
```
sudo apt update && sudo apt install p7zip-full
```
Create a Symbolic Link
```
sudo ln -s /usr/bin/7z /usr/bin/7zz
```
I solved the problem like this

devttys0, 2025-04-05:
Should be resolved with [this](https://github.com/ReFirmLabs/binwalk/pull/844) PR.

closed 2025-04-05 by devttys0