ReFirmLabs/binwalk #841: Jffs2 folder extracted empty

Vincent964, 2025-03-17:
Hi, I am trying to extract my configuration from my router. I have the firmware.bin extracted from it. With binwalk I can extract the squash filesystem perfectly while for jffs2 I get an empty folder. The Jefferson tool is already installed, how come I get the empty folder? Thanks regards

devttys0, 2025-03-17:
What happens when you run `jefferson` against the JFFS2 image manually? Can you provide the sample firmware image that you're running binwalk against?

Vincent964, 2025-03-17:
Hi thanks for your reply, I tried many different versions of `jefferson`, up to 0.4.6 which is the latest. (i used `dd` to extract only jffs2) then I run:
```
jefferson -d jffs2-root -vvv jffs.bin

dumping fs to /home/vincent/jefferson/outputjffs (endianness: >)
Jffs2_raw_inode count: 0
Jffs2_raw_dirent count: 0
```

But unfortunately the jffs2-root folder still remains empty.
I can send you the entire complete bin file `fw.bin` that I extracted from my router, but I would prefer to share it with you privately if possible, Thanks regards

Vincent964, 2025-03-18:
Please if you could do this for me I would be grateful, I need to recover my device configuration and this is the only way. Best regards

Vincent964, 2025-04-03:
> What happens when you run `jefferson` against the JFFS2 image manually? Can you provide the sample firmware image that you're running binwalk against?

Please if you could do this for me I would be grateful, I need to recover my device configuration and this is the only way. Best regards

devttys0, 2025-04-05:
If there are in fact any files contained in the JFFS2 image, then this is a bug in [jefferson](https://github.com/sviehb/jefferson)

closed 2025-04-05 by devttys0

Vincent964, 2025-04-08:
Hi, thanks for the reply, sure but my doubt instead was if maybe what binwalk recognizes as a jffs2 file is not actually. Or it is recognized but the layout is fragmented or raw. Can you help me with this?