ReFirmLabs/binwalk #852: Request to reopen issue #841 – JFFS2 extraction problem still present

Vincent964, 2025-04-11:
Hi,
I noticed that issue #841 was closed, but the problem is still present.
I’ve also tested the extracted dump from other working Ubiquiti devices I own, and I consistently get an empty folder from binwalk when trying to extract the JFFS2 partition.

It seems the behavior is still occurring, even under different conditions.
If possible, I kindly ask you to reopen the issue or let me know if you’d prefer me to open a new one with more details.

Thanks for the work you’re doing on the project!

devttys0, 2025-04-12:
This issue may well be a false positive in binwalk's part, but without a sample file for testing there is no way to verify the problem or fix it.

Vincent964, 2025-04-12:
Hi,
as I mentioned in the previous issue, I’m able to provide a sample file for testing — however, I cannot share it publicly.
If there’s a private channel (email or other) where I can send the file securely, I’d be happy to share it for analysis.

Let me know what works best for you, and thanks again for your precious help

devttys0, 2025-04-12:
You can email them to: heffnercj@gmail.com

Vincent964, 2025-04-12:
Ok email sent, thanks

devttys0, 2025-04-14:
I took a look at the reported JFFS2 image in the firmware; it is a valid JFFS2 image, it's just empty:

```
00000000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00000010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00010000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00010010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00020000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00020010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00030000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00030010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00040000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00040010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00050000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00050010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00060000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00060010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00070000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00070010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00080000  19 85 20 03 00 00 00 0c  f0 60 dc 98 ff ff ff ff  |.. ......`......|
00080010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00090000  19 85 20 03 00 00 00 0c  f0 60 dc 98              |.. ......`..|
0009000c
```

There is a valid JFFS2 node every `0x10000` bytes, each node starts with the expected JFFS2 magic bytes (`19 85`) and has a valid checksum `f0 60 dc 98`. However, each node's reported size (`0x0000000C`) is simply the size of the JFFS2 node header and the nodes contain no data.

So while this is a valid JFFS2 file system, there is no data to be extracted here.

closed 2025-04-14 by devttys0