ReFirmLabs/binwalk #853: cargo install binwalk failing

mubix, 2025-04-11:
This is on Debian Bookworm (12), Cargo version 1.65.0

``` 
Updating crates.io index
  Downloaded binwalk v3.1.0
  Downloaded 1 crate (562.6 KB) in 0.38s
  Installing binwalk v3.1.0
error: failed to compile `binwalk v3.1.0`, intermediate artifacts can be found at `/tmp/cargo-installtQOYcI`

Caused by:
  failed to select a version for `env_logger`.
      ... required by package `binwalk v3.1.0`
  versions that meet the requirements `^0.11.5` are: 0.11.8, 0.11.7, 0.11.6, 0.11.5

  the package `binwalk` depends on `env_logger`, with features: `anstream` but `env_logger` does not have these features.
   It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.

  failed to select a version for `env_logger` which could resolve this conflict
```

amaturo78, 2025-06-21:
Hi, unfortunately same problem :)

debian12 6.1.0-27-amd64
cargo 1.87.0
rustc 1.87.0

cd /home/binwalk
sudo cargo update

Updating crates.io index
    Updating git repository `https://github.com/devttys0/delink`
error: failed to select a version for `env_logger`.
    ... required by package `binwalk v3.1.1 (/home/binwalk)`
versions that meet the requirements `^0.11.5` are: 0.11.8, 0.11.7, 0.11.6, 0.11.5

the package `binwalk` depends on `env_logger`, with features: `anstream` but `env_logger` does not have these features.
 It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.

failed to select a version for `env_logger` which could resolve this conflict

Suggestions?

amaturo78, 2025-06-21:
Sort it out !

sudo nano Cargo.toml
env_logger = { version = "0.11", default-features = false, features = ["color"] }
cargo build --release