ReFirmLabs/binwalk #854: Binwalk not detecting UEFI signatures

deliciouslytyped, 2025-04-13:
I have a sample that starts with the following:
```
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000010: d9 54 93 7a 68 04 4a 44 81 ce 0b f6 17 d8 90 df  .T.zh.JD........
00000020: 00 00 01 00 00 00 00 00 5f 46 56 48 ff 8e ff ff  ........_FVH....
00000030: 48 00 b1 4c 00 00 00 01 01 00 00 00 00 00 01 00  H..L............
```

Binwalk 2.3.3 detects a UEFI PI header, but binwalk 3.1.0 does not.

```
$ binwalk bdump

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             UEFI PI Firmware Volume, volume size: 65536, header size: 0, revision: 0, EFI Firmware File System, GUID: 7A9354D9-0468-444A-CE81-0BF617D890DF
...
```

```
$ binwalk -e bdump

                                                      bdump
--------------------------------------------------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
--------------------------------------------------------------------------------------------------------------------------------------------------------
539818                             0x83C0C                            Windows PE binary, machine type: Intel x86
...
```

devttys0, 2025-04-14:
I've pushed a change that I think will fix this, though without the binary to test against I don't know for sure. Let me know if it works! :)

devttys0, 2025-04-17:
Binwalkv3 has generally much stricter validation than previous versions. In the case of UEFI, the version field was expected to be `2` as that is what I found in all docs/code/samples that I looked at. In the hexdump you provided, the version field is set to `1`.

closed 2025-07-08 by devttys0