ReFirmLabs/binwalk #859: No output MacOS

McFacePunch, 2025-04-29:
When running the homebrew version of binwalk on arm64 Mac running sequoia there is no output.

[image]

m-m-adams, 2025-05-28:
Same problem here, running it on a file that I know has a squashfs system and it's not being detected on mac

devttys0, 2025-07-08:
@McFacePunch  That is the expected output if no valid signatures were detected in the provided file. If you think this is in error, please provide the files in question.

@m-m-adams  SquashFS should definitely be detected. Please provide the file in question so I can track down the bug.

m-m-adams, 2025-07-08:
@devttys0 I've since noticed it's also present on linux so I'll open a new issue