ReFirmLabs/binwalk #862: v3.1.0 Missing arbitrary file extraction

0bs3n, 2025-05-07:
As far as I can tell, the binwalk re-write has lost the ability extract arbitrary files (i.e. if I have a bunch of ELF files baked into a firmware image), which used to be possible via the `--dd` flag, specifying a regex for matching and a file suffix for the output files.

devttys0, 2025-05-20:
Try the `--carve` option. If you only want to pull out the ELF files, you can combine this with the `--include` option:

```
binwalk --carve --include=elf some_file.bin
```

hauklandsverk2, 2025-05-26:
telepromtmac@HauksMac derror: unexpected argument '--carve' found

  tip: to pass '--carve' as a value, use '-- --carve'

Usage: binwalk [OPTIONS] [FILE_NAME]

For more information, try '--help'.

devttys0, 2025-07-08:
Yes, the `--carve` option was not added until after the 3.1.0 release. You'll need to grab the latest code from the master branch.

closed 2025-07-08 by devttys0