ReFirmLabs/binwalk #881: Binwalk 3.1.0 fails to detect squashfs and U Boot partitions detected by binwalk 2.3.4

m-m-adams, 2025-07-08:
[image]

Binwalk 3.1.0 fails to identify u boot and squashfs in the attached firmware image from a stock tplink router. Running with the `-a` option only finds a "GPG signed file" that I believe is a false positive

[image]

This shows binwalk 2.3.4 correctly identifying both U boot and squashfs in the same image

[firmware.zip](https://github.com/user-attachments/files/21130764/firmware.zip)

devttys0, 2025-07-09:
IIRC, the U-Boot version string signature was not included in the 3.1.0 release. It is in the current master branch and correctly reports the U-Boot version.

While there is a SquashFS header at offset `0x100000`, it appears to be invalid which is why Binwalk 3.1.0 is not reporting it. Specifically, the UID table is said to exist at a file offset of `0xffffffffffffffff`, and the usual SquashFS tools report that it is corrupted:

```
$ sasquatch test.img 
SquashFS version [4.0] / inode count [625] suggests a SquashFS image of the same endianess
Lseek failed because Invalid argument
read_block: failed to read block @0xffffffffffffffff
read_id_table: failed to read id table block
FATAL ERROR: File system corruption detected

$ unsquashfs test.img 
Lseek failed because Invalid argument
read_block: failed to read block @0xffffffffffffffff
read_id_table: failed to read id table block
FATAL ERROR: File system corruption detected
```

This could be due to a corrupt firmware image, or it could be a vendor-specific modification. Can you provide more information about the firmware image? Was it downloaded from a vendor's website, or is it a raw flash dump off a physical device?

m-m-adams, 2025-07-09:
It's a dump from the physical memory. I have several of these routers to teach a course on hardware hacking and others do extract fine with the current binwalk so there's possibly just something wrong with this one. Would it be possible to add some sort of error message in this case?

closed 2025-08-11 by devttys0