ReFirmLabs/binwalk #896: bug `binwalk -Me firmware`

jelasin, 2025-08-08:
➜  linksys binwalk -Me ./FW_FGMM601-HK_1.1.25.000001_prod.bin
[2025-08-08T03:33:12Z ERROR binwalk::binwalk] Failed to create symlink /mnt/f/linksys/extractions/FW_FGMM601-HK_1.1.25.000001_prod.bin -> /mnt/f/linksys/FW_FGMM601-HK_1.1.25.000001_prod.bin: File exists (os error 17)

thread 'main' panicked at /home/ub24/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/binwalk-3.1.0/src/main.rs:100:6:
Binwalk initialization failed: BinwalkError
stack backtrace:
   0:     0x617383303942 - <std::sys::backtrace::BacktraceLock::print::DisplayBacktrace as core::fmt::Display>::fmt::hf435e8e9347709a8
   1:     0x61738332ac23 - core::fmt::write::h0a51fad3804c5e7c
   2:     0x617383300e73 - std::io::Write::write_fmt::h9759e4151bf4a45e
   3:     0x617383303792 - std::sys::backtrace::BacktraceLock::print::h1ec5ce5bb8ee285e
   4:     0x6173833054dc - std::panicking::default_hook::{{closure}}::h5ffefe997a3c75e4
   5:     0x6173833052df - std::panicking::default_hook::h820c77ba0601d6bb
   6:     0x617383305e72 - std::panicking::rust_panic_with_hook::h8b29cbe181d50030
   7:     0x617383305c2a - std::panicking::begin_panic_handler::{{closure}}::h9f5b6f6dc6fde83e
   8:     0x617383303e49 - std::sys::backtrace::__rust_end_short_backtrace::hd7b0c344383b0b61
   9:     0x6173833058bd - __rustc[5224e6b81cd82a8f]::rust_begin_unwind
  10:     0x617383069a40 - core::panicking::panic_fmt::hc49fc28484033487
  11:     0x617383069e06 - core::result::unwrap_failed::h9e4c136384b1cfa3
  12:     0x6173830d3823 - binwalk::main::hecd5f524dc273ee1
  13:     0x61738311dd43 - std::sys::backtrace::__rust_begin_short_backtrace::h6e88424403699e01
  14:     0x6173830c7a99 - std::rt::lang_start::{{closure}}::h3fcf187187749ab1
  15:     0x6173832f82e0 - std::rt::lang_start_internal::h31bbb7f936fd6b5d
  16:     0x6173830d9dd5 - main
  17:     0x70f84982a1ca - __libc_start_call_main
                               at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  18:     0x70f84982a28b - __libc_start_main_impl
                               at ./csu/../csu/libc-start.c:360:3
  19:     0x61738306a1d5 - _start
  20:                0x0 - <unknown>

jelasin, 2025-08-08:
move it in ~/
➜  linksys pwd
/home/ub24/linksys
➜  linksys binwalk -Me FW_FGMM601-HK_1.1.25.000001_prod.bin
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to execute command dtc["-I", "dtb", "-O", "dts", "-o", "system.dtb", "/home/ub24/linksys/extractions/FW_FGMM601-HK_1.1.25.000001_prod.bin.extracted/96BE5B/dtb_96BE5B.dtb"]: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to spawn external extractor for 'dtb' signature: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to execute command dtc["-I", "dtb", "-O", "dts", "-o", "system.dtb", "/home/ub24/linksys/extractions/FW_FGMM601-HK_1.1.25.000001_prod.bin.extracted/96BE5B/dtb_96BE5B.dtb"]: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to spawn external extractor for 'dtb' signature: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to execute command dtc["-I", "dtb", "-O", "dts", "-o", "system.dtb", "/home/ub24/linksys/extractions/FW_FGMM601-HK_1.1.25.000001_prod.bin.extracted/96ED47/dtb_96ED47.dtb"]: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to spawn external extractor for 'dtb' signature: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to execute command dtc["-I", "dtb", "-O", "dts", "-o", "system.dtb", "/home/ub24/linksys/extractions/FW_FGMM601-HK_1.1.25.000001_prod.bin.extracted/96ED47/dtb_96ED47.dtb"]: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to spawn external extractor for 'dtb' signature: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to execute command dtc["-I", "dtb", "-O", "dts", "-o", "system.dtb", "/home/ub24/linksys/extractions/FW_FGMM601-HK_1.1.25.000001_prod.bin.extracted/9754A9/dtb_9754A9.dtb"]: No such file or directory (os error 2)
[2025-08-08T03:37:16Z ERROR binwalk::extractors::common] Failed to spawn external extractor for 'dtb' signature: No such file or directory (os error 2)
[2025-08-08T03:37:17Z ERROR binwalk::extractors::common] Failed to execute command dtc["-I", "dtb", "-O", "dts", "-o", "system.dtb", "/home/ub24/linksys/extractions/FW_FGMM601-HK_1.1.25.000001_prod.bin.extracted/9754A9/dtb_9754A9.dtb"]: No such file or directory (os error 2)

devttys0, 2025-08-11:
The first `Failed to create symlink` error has been fixed in the latest master branch. Originally Binwalkv3 would refuse to extract the same file name to the same output directory to prevent overwriting previous work. It will now overwrite previous extractions and this error message should not occur.

The second `No target file name specified! Try --help.` should be self-explanatory, though in the master branch this error message has been made more user-friendly.

The `Failed to execute command dtc...No such file or directory` indicates that you do not have the `dtc` utility installed. On Ubuntu this can be installed with `sudo apt-get install device-tree-compiler`; other Linux distros should have a similar package. You may want to look at the dependencies scripts for installing run-time dependencies, or look at the output of `binwalk --list` which lists all available signatures and the external command used for extraction of each signature.

closed 2025-08-11 by devttys0