ReFirmLabs/binwalk #902: Feature request: Support for larger-than-available-memory files/devices by seeking through the file instead of attempting to load the entire thing into memory?

ipaqmaster, 2025-08-28:
Is this a reasonable feature request? I'd love if binwalk could go over large files and images without causing the host to run out of memory as it tries to load them entirely into it

CFSworks, 2025-09-10:
I have also been affected by this. Accessing the file through `mmap()` may be a good option: it still makes the file available in the memory space of the program as expected by current design, but it allows the operating system to page the target on demand.

okh-mzny, 2025-12-06:
This limitation stems from the use of Rust, where `mmap()`ing a file is considered an unsafe practice, as the file could be modified outside the control of the borrow checker.

There is a PR open (https://github.com/ReFirmLabs/binwalk/pull/762) which tries to add a classic memory saving `mmap()`, however there seem to be some issues that will need to be addressed with it